ESP32Async/ESPAsyncWebServer #21: [BUG] server not reachable after calling end() and begin().

mkalkbrenner, 2025-01-27:
- Which board: ESP32S3 

```cpp
AsyncWebServer *server = new AsyncWebServer(80);
// Start the web server
server->begin();

delay(10000);
server->end();

delay(10000);
server->begin();
```

The server works fine. But if I stop the sever and start it again later, it isn't reachable anymore.

mkalkbrenner, 2025-01-27:
Some more tests.
Even that doesn't work:
```cpp
AsyncWebServer *server = new AsyncWebServer(80);
// Start the web server
server->begin();

delay(10000);
server->end();
delete server;
server = nullptr;

delay(10000);
server = new AsyncWebServer(80);
server->begin();
```

The server is not reachable on port 80 after the "restart".

But if I do this, I can reach the server on port 81:
```cpp
AsyncWebServer *server = new AsyncWebServer(80);
// Start the web server
server->begin();

delay(10000);
server->end();
delete server;
server = nullptr;

delay(10000);
server = new AsyncWebServer(81);
server->begin();
```

mathieucarbou, 2025-01-27:
I can reproduce indeed.