ESP32Async/ESPAsyncWebServer #47: Allow duplicate headers in response according to RFC

mathieucarbou, 2025-02-02:
### Discussed in https://github.com/ESP32Async/ESPAsyncWebServer/discussions/46

<div type='discussions-op-text'>

<sup>Originally posted by **KarolBychowski** February  2, 2025</sup>
Only one cookie is always added, the last one (image). And this is all because in the WebResponses.cpp file there is a check if the set-cookie already exists, if so, we remove it and add a new one. If we comment out "return false", then everything works fine. (image)

You need to ask for false to setResponse->addHeader("Set-Cookie", "SessionID=" + sessionID, **false**)

[image]
[image]
</div>

DRSDavidSoft, 2025-02-02:
@mathieucarbou If you recall, this is exactly what we originally discussed in https://github.com/mathieucarbou/ESPAsyncWebServer/pull/82#issuecomment-2324914509, hence the "breaking change" warning.

I believe @me-no-dev's approach in #48 is a good way to implement it, i.e. to keep track of what headers are allowed once vs. allowed multiple times, per RFC convention.

It's very important to document this in the library. What happens when a new header with the same key is requested by the user to be added:

1. If no such header exists, add it to the response
2. If a header with the same key already exists:
  a. Can multiple values for the same key exist? Then add it to the response (i.e. both will be kept)
  b. If `replaceExisting` set to false, the new value is ignored, i.e. the old value will be kept
  c. otherwise, the old one is removed, and the new one will take its place

[image]

mathieucarbou, 2025-02-02:
I love your graph! It would be nice to PR that!

DRSDavidSoft, 2025-02-02:
@mathieucarbou Sure! I was also thinking about making a graph for the request flow, so that it would make clear to the user about the sequence of when the handlers, middlewares, rewrites, etc. all take effect into a request.

Since this is off-topic to this particular issue, I'd appreciate it if you could make a new branch and a discussion to coordinate reworking the documentation and adding flows, where required. I'd love to contribute with adding some flows, and maybe even organizing existing docs to a level 👍🏻

mathieucarbou, 2025-02-02:
@DRSDavidSoft : could be PR-ed in the readme fyi if you want

closed 2025-02-02 by mathieucarbou