ESP32Async/ESPAsyncWebServer #71: Issue with handleUpload

TheMechanos, 2025-02-12:
### Platform

ESP32

### IDE / Tooling

PlatformIO

### What happened?

I'm experiencing an issue with handling file uploads in the Async Web Server on ESP32. The handleUpload function is called only once and receives only the first chunk of data.

For small files (that fit within a single chunk), the upload works correctly. However, larger files that are sent in multiple chunks fail, as handleUpload is not called for subsequent parts.

When I use latest https://github.com/me-no-dev/ESPAsyncWebServer problem don't exists.

### Stack Trace

-

### Minimal Reproductible Example (MRE)

Board: ESP32-S3

```
void handleUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
    if (!index) {
        Serial.printf("UploadStart: %s\n", filename.c_str());
    }
    if(len){
        Serial.printf("Uploading %u B\n", len);
    }
    if (final) {
        Serial.printf("UploadEnd: %s, %u B\n", filename.c_str(), index + len);
    }
}

void serverConfig(){
    server.on(
        "/api/file", HTTP_POST,
        [](AsyncWebServerRequest *request) {

            request->send(200);
        },
        handleUpload);

    server.begin();
    Serial.println("Server init done!");
}
```

Log:
```
08:35:11.487 > UploadStart: sample-3s.mp3
08:35:11.508 > Uploading 1436 B
```
And this is all, no more chunks.

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

mathieucarbou, 2025-02-12:
I am able to reproduce with latest version... Looking...
Could you please try to rollback to 3.6.0 ? Does it works ?

TheMechanos, 2025-02-12:
I do a test on this repo:

3.7.0 - Bug described in this issue
3.6.2 - Working good