ESP32Async/ESPAsyncWebServer #76: ArduinoJSON functions are not compiled

controlol, 2025-02-13:
### Platform

ESP32

### IDE / Tooling

ESP-IDF

### What happened?

After updating to this repo there seem to be some issues with compiling the AsyncJson.cpp files. In the header (AsyncJson.h) ArduinoJSON can be included and therefor the header files are correctly generated, but in the cpp file `ASYNC_JSON_SUPPORT` is not defined. This results in linker errors thrown, provided in the stack trace.

If the idf_component.yml file is updated to also have bblanchon/arduinojson as a dependency the file is compiled as expected and the build succeeds.

### Stack Trace

C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj):(.literal._ZN12HeroBalancer8LocalApi11LocalWebApi23_handlePreflightRequestEP21AsyncWebServerRequest+0x0): undefined reference to `AsyncJsonResponse::AsyncJsonResponse(bool)'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj):(.literal._ZN12HeroBalancer8LocalApi11LocalWebApi23_handlePreflightRequestEP21AsyncWebServerRequest+0x4): undefined reference to `AsyncJsonResponse::setLength()'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj):(.literal._ZN12HeroBalancer8LocalApi11LocalWebApi4postEPKcSt8functionIF12t_http_codesP21AsyncWebServerRequestRN11ArduinoJson8V730PB2211JsonVariantEP17AsyncJsonResponseEE+0x4): undefined reference to `AsyncCallbackJsonWebHandler::AsyncCallbackJsonWebHandler(String const&, std::function<void (AsyncWebServerRequest*, ArduinoJson::V730PB22::JsonVariant&)>)'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj): in function `HeroBalancer::LocalApi::LocalWebApi::_handlePreflightRequest(AsyncWebServerRequest*)':
C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:110:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi23_handlePreflightRequestEP21AsyncWebServerRequest+0x12): undefined reference to `AsyncJsonResponse::AsyncJsonResponse(bool)'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:115:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi23_handlePreflightRequestEP21AsyncWebServerRequest+0x32): undefined reference to `AsyncJsonResponse::setLength()'   
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj): in function `HeroBalancer::LocalApi::LocalWebApi::_handlePostRequest(AsyncWebServerRequest*, ArduinoJson::V730PB22::JsonVariant&, std::function<t_http_codes (AsyncWebServerRequest*, ArduinoJson::V730PB22::JsonVariant&, AsyncJsonResponse*)>&)':
C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:64:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi18_handlePostRequestEP21AsyncWebServerRequestRN11ArduinoJson8V730PB2211JsonVariantERSt8functionIF12t_http_codesS3_S7_P17AsyncJsonResponseEE+0x10): undefined reference to `AsyncJsonResponse::AsyncJsonResponse(bool)'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:73:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi18_handlePostRequestEP21AsyncWebServerRequestRN11ArduinoJson8V730PB2211JsonVariantERSt8functionIF12t_http_codesS3_S7_P17AsyncJsonResponseEE+0x4d): undefined reference to `AsyncJsonResponse::setLength()'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj): in function `HeroBalancer::LocalApi::LocalWebApi::_handleGetRequest(AsyncWebServerRequest*, std::function<t_http_codes (AsyncWebServerRequest*, AsyncJsonResponse*)>&)':
C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:82:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi17_handleGetRequestEP21AsyncWebServerRequestRSt8functionIF12t_http_codesS3_P17AsyncJsonResponseEE+0x10): undefined reference to `AsyncJsonResponse::AsyncJsonResponse(bool)'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:91:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi17_handleGetRequestEP21AsyncWebServerRequestRSt8functionIF12t_http_codesS3_P17AsyncJsonResponseEE+0x4d): undefined reference to `AsyncJsonResponse::setLength()'
C:/Espressif/tools/xtensa-esp-elf/esp-13.2.0_20240530/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld.exe: esp-idf/main/libmain.a(LocalWebApi.cpp.obj): in function `HeroBalancer::LocalApi::LocalWebApi::post(char const*, std::function<t_http_codes (AsyncWebServerRequest*, ArduinoJson::V730PB22::JsonVariant&, AsyncJsonResponse*)>)':
C:/Users/Luc/websites/HeroBalancer/HeroGatewayFirmware-V4/main/api/LocalWebApi.cpp:51:(.text._ZN12HeroBalancer8LocalApi11LocalWebApi4postEPKcSt8functionIF12t_http_codesP21AsyncWebServerRequestRN11ArduinoJson8V730PB2211JsonVariantEP17AsyncJsonResponseEE+0xde): undefined reference to `AsyncCallbackJsonWebHandler::AsyncCallbackJsonWebHandler(String const&, std::function<void (AsyncWebServerRequest*, ArduinoJson::V730PB22::JsonVariant&)>)'
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.

### Minimal Reproductible Example (MRE)

Used versions:
espressif/idf: 5.3.2
espressif/arduino-esp32: 3.1.1
bblanchon/arduinojson: 7.3.0
esp32async/asynctcp: 3.3.5
esp32async/ESPAsyncWebServer: 3.7.0

Espressif Registry is used to install the dependencies

Using any function from AsyncJson.h won't work and result in linker errors. 

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

controlol, 2025-02-13:
I tried to add the full macro block from the header file to the cpp file and that made no differences (duh, that should already be included).

https://github.com/ESP32Async/ESPAsyncWebServer/blob/dbf498e5fb4ff16a12636eec33c6db579e8cf43b/src/AsyncJson.h#L7-L14

Here are the includes where AsyncJson is used:

```c
#include <AsyncJson.h>
#include <ArduinoJson.h>
#include <HTTPClient.h>
```

I also tried adding all the includes similar to the example.

controlol, 2025-02-13:
I wanted to see if it would be possible to link a dependency to the component from the main CMakeLists.txt. This could be accomplished using [Adding Link-Time Dependencies](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#adding-link-time-dependencies) or simply with CMake [target_link_libraries](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-guides/build-system.html#advanced-workaround-undefined-symbols).

Adding the following to main/CMakeLists.txt produced a successful build for me:

```Cmake
target_link_libraries(__idf_esp32async__espasyncwebserver PRIVATE __idf_bblanchon__arduinojson)
```

I am not sure if there's a pio equivalent for that.

me-no-dev, 2025-02-13:
fix incoming: https://github.com/ESP32Async/ESPAsyncWebServer/pull/78

closed 2025-02-13 by mathieucarbou