ESP32Async/ESPAsyncWebServer #105: Does 3.6 -> HEAD do not build for an esp32-c3

ieb, 2025-03-05:
### Platform

ESP32

### IDE / Tooling

PlatformIO

### What happened?

Code that built previously with a me-no-dev version from Oct 2024 now fails to buid with all versions from 3.6.0 through to the latest release.

### Stack Trace

                    In file included from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/riscv32-esp-elf/rv32imc/ilp32/no-rtti/bits/c++allocator.h:33,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/allocator.h:46,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/string:41,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/stdexcept:39,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/array:39,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/tuple:39,
                                     from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/functional:54,
                                     from /Users/ieb/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiGeneric.h:28,
                                     from /Users/ieb/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiSTA.h:28,
                                     from /Users/ieb/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:32,
                                     from lib/network/network.h:3,
                                     from lib/network/echoserver.cpp:1:
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/ext/new_allocator.h: In instantiation of 'void __gnu_cxx::new_allocator<_Tp>::construct(_Up*, _Args&& ...) [with _Up = std::unique_ptr<AsyncWebHandler>; _Args = {const std::unique_ptr<AsyncWebHandler, std::default_delete<AsyncWebHandler> >&}; _Tp = std::_List_node<std::unique_ptr<AsyncWebHandler> >]':
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/alloc_traits.h:475:4:   required from 'static void std::allocator_traits<std::allocator<_CharT> >::construct(std::allocator_traits<std::allocator<_CharT> >::allocator_type&, _Up*, _Args&& ...) [with _Up = std::unique_ptr<AsyncWebHandler>; _Args = {const std::unique_ptr<AsyncWebHandler, std::default_delete<AsyncWebHandler> >&}; _Tp = std::_List_node<std::unique_ptr<AsyncWebHandler> >; std::allocator_traits<std::allocator<_CharT> >::allocator_type = std::allocator<std::_List_node<std::unique_ptr<AsyncWebHandler> > >]'
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_list.h:645:33:   required from 'std::__cxx11::list<_Tp, _Alloc>::_Node* std::__cxx11::list<_Tp, _Alloc>::_M_create_node(_Args&& ...) [with _Args = {const std::unique_ptr<AsyncWebHandler, std::default_delete<AsyncWebHandler> >&}; _Tp = std::unique_ptr<AsyncWebHandler>; _Alloc = std::allocator<std::unique_ptr<AsyncWebHandler> >; std::__cxx11::list<_Tp, _Alloc>::_Node = std::_List_node<std::unique_ptr<AsyncWebHandler> >]'
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_list.h:1903:10:   required from 'void std::__cxx11::list<_Tp, _Alloc>::_M_insert(std::__cxx11::list<_Tp, _Alloc>::iterator, _Args&& ...) [with _Args = {const std::unique_ptr<AsyncWebHandler, std::default_delete<AsyncWebHandler> >&}; _Tp = std::unique_ptr<AsyncWebHandler>; _Alloc = std::allocator<std::unique_ptr<AsyncWebHandler> >; std::__cxx11::list<_Tp, _Alloc>::iterator = std::_List_iterator<std::unique_ptr<AsyncWebHandler> >]'
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_list.h:1235:4:   required from 'void std::__cxx11::list<_Tp, _Alloc>::emplace_back(_Args&& ...) [with _Args = {const std::unique_ptr<AsyncWebHandler, std::default_delete<AsyncWebHandler> >&}; _Tp = std::unique_ptr<AsyncWebHandler>; _Alloc = std::allocator<std::unique_ptr<AsyncWebHandler> >]'
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_list.h:1832:6:   required from 'void std::__cxx11::list<_Tp, _Alloc>::_M_initialize_dispatch(_InputIterator, _InputIterator, std::__false_type) [with _InputIterator = std::_List_const_iterator<std::unique_ptr<AsyncWebHandler> >; _Tp = std::unique_ptr<AsyncWebHandler>; _Alloc = std::allocator<std::unique_ptr<AsyncWebHandler> >]'
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/stl_list.h:746:9:   required from 'std::__cxx11::list<_Tp, _Alloc>::list(const std::__cxx11::list<_Tp, _Alloc>&) [with _Tp = std::unique_ptr<AsyncWebHandler>; _Alloc = std::allocator<std::unique_ptr<AsyncWebHandler> >]'
                    .pio/libdeps/esp32-c3/ESPAsyncWebServer/src/ESPAsyncWebServer.h:1064:7:   required from here
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/ext/new_allocator.h:136:4: error: use of deleted function 'std::unique_ptr<_Tp, _Dp>::unique_ptr(const std::unique_ptr<_Tp, _Dp>&) [with _Tp = AsyncWebHandler; _Dp = std::default_delete<AsyncWebHandler>]'
                      { ::new((void *)__p) _Up(std::forward<_Args>(__args)...); }
                        ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
                    In file included from /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/memory:80,
                                     from /Users/ieb/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFiClient.h:26,
                                     from /Users/ieb/.platformio/packages/framework-arduinoespressif32/libraries/WiFi/src/WiFi.h:37,
                                     from lib/network/network.h:3,
                                     from lib/network/echoserver.cpp:1:
                    /Users/ieb/.platformio/packages/toolchain-riscv32-esp/riscv32-esp-elf/include/c++/8.4.0/bits/unique_ptr.h:397:7: note: declared here
                           unique_ptr(const unique_ptr&) = delete;

### Minimal Reproductible Example (MRE)

Anything using the following seems to cause the problem.
Removing all references to the ESPAsyncWebServer from the codebase and removing the dependencies eliminates the build failure.

            [env:esp32-c3]
            platform = espressif32
            board = esp32-c3-devkitm-1
            framework = arduino
            board_build.mcu = esp32c3
            board_build.f_cpu = 160000000L
            build_flags = 
                -DASYNCWEBSERVER_REGEX
                -D ARDUINO_USB_MODE=1
                -D ARDUINO_USB_CDC_ON_BOOT=1 
                !echo '#define GIT_SHA1_VERSION "'$(git log |head -1 |cut -c8-)'"' >> src/version.h
            lib_deps = 
                https://github.com/ttlappalainen/NMEA2000.git
                https://github.com/ttlappalainen/NMEA0183.git
                https://github.com/ieb/NMEA2000_twai.git#platformIOUse
                ESP32Async/AsyncTCP
                ESP32Async/ESPAsyncWebServer
                paulstoffregen/Time@^1.6.1
                ModbusClient=https://github.com/eModbus/eModbus.git

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

ieb, 2025-03-05:
Please dont do anything yet. I am double checking its not me.

mathieucarbou, 2025-03-05:
We do build on C3 and build pass: https://github.com/ESP32Async/ESPAsyncWebServer/actions/runs/13674993269/job/38233414420

that is not an issue - will convert it to support discussion.