ESP32Async/ESPAsyncWebServer #139: Regression in AsyncWebServerResponse beginResponse

mozgy, 2025-03-30:
### Platform

ESP32

### IDE / Tooling

pioarduino

### What happened?

This worked fine before merge (I've used me-no-dev's code) ->

```
  photoFrameBuffer = esp_camera_fb_get();
  photoFrameLength = photoFrameBuffer->len;
  photoFrame += String( (char *) photoFrameBuffer->buf, photoFrameLength );
  AsyncWebServerResponse *response = request->beginResponse( 200, "image/jpeg", photoFrame );
  request->send( response );

```

Now on the web client side I only get <FF><D8><FF><E0> ..

### Stack Trace

PACKAGES:
 - framework-arduinoespressif32 @ 3.2.0
 - framework-arduinoespressif32-libs @ 5.4.0+sha.2f7dcd862a

Dependency Graph
|-- ESPAsyncWebServer @ 3.7.4
|-- ArduinoJson @ 7.1.0
|-- AsyncTCP @ 3.3.8

### Minimal Reproductible Example (MRE)

```
  photoFrameBuffer = esp_camera_fb_get();
  photoFrameLength = photoFrameBuffer->len;
  photoFrame += String( (char *) photoFrameBuffer->buf, photoFrameLength );
  AsyncWebServerResponse *response = request->beginResponse( 200, "image/jpeg", photoFrame );
  request->send( response );

```

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

mozgy, 2025-03-30:
.. missed proper copy/paste ->

Now on the web client side I only get <FF><D8><FF><E0>

mathieucarbou, 2025-03-30:
@mozgy : please explain what is wrong. Paste compile error or stack trace. thanks!