ESP32Async/ESPAsyncWebServer #242: Lost writes using AsyncCallbackResponse

jonny5532, 2025-07-28:
### Platform

ESP32

### IDE / Tooling

Arduino (IDE/CLI)

### What happened?

When sending a 16kb response in three parts, using `AsyncCallbackResponse`, the middle one sometimes goes missing (so the received data is ~5k shorter than expected - the page finishes but with ~5k fewer bytes than the `Content-Type` indicated).

This seems to be due to this:

https://github.com/ESP32Async/ESPAsyncWebServer/blob/80af24527446dff31476d0e432e436e030b4249d/src/WebResponses.cpp#L476

where the code assumes that the whole buffer got written (it `free`s it after), even if it was only partially sent. Ideally it could signal to the `AsyncCallbackResponse` that it needs to rewind, but being in the Abstract superclass it can't access `_filledLength` to decrement it.

AsyncTCP's `write(...)` allocates memory by default, and this system is very memory constrained.

The best fix is probably to store the unwritten data somewhere temporarily, but it can't use the cache as that's only for template responses.  Or add a `rewind` method which the `AsyncCallbackResponse` can implement to roll back `_filledLength` (I've tried this locally, and it works, but this might affect `AsyncStreamResponse` too (unsure) and they can't be rewound).

Unless I'm missing something? Thanks!

### Stack Trace

N/A

### Minimal Reproductible Example (MRE)

This is the essence of it, but could build a full MRE if needed?
```C++
String *largeResponseString = new String("16k chars...");

AsyncWebServerResponse *response = request->beginResponse(
  "text/html",
  largeResponseString->length(),
  [](uint8_t *buffer, size_t maxLen, size_t alreadySent) -> size_t {
    // Calculate how much we can send in this chunk
    size_t remaining = largeResponseString->length() - alreadySent;
    size_t toSend = (remaining < maxLen) ? remaining : maxLen;

    // Copy the data from the content string to the buffer
    memcpy(buffer, largeResponseString->c_str() + alreadySent, toSend);
    // Return the number of bytes sent
    return toSend;
  },
  (AwsTemplateProcessor)nullptr  // No template processor needed
);
request->send(response);
```

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

willmmiles, 2025-07-28:
> CC [@willmmiles](https://github.com/willmmiles)

@mathieucarbou You pegged me all right: [I did indeed solve this one in the WLED fork.](https://github.com/Aircoookie/ESPAsyncWebServer/blob/2d3af2c32b1dd1a8fd9fb01d7b51fa3653221a89/src/WebResponses.cpp#L416)   Graceful handling of low memory conditions was the overarching goal behind the changes in that project.  You can build a test case by consuming internal DRAM such that `client()->space() > heap_caps_get_largest_free_block(MALLOC_CAP_INTERNAL | MALLOC_CAP_DEFAULT)` - essentially arranging that there isn't enough heap available to allocate the full 2*MSS that the TCP connection could permit.

The solution there ended up being fairly complex:
- First, keep the output buffer around if it doesn't send completely; track how much of it is left
- Second, if there's leftover content in the output buffer at the start of an `_ack` call, send it first before considering anything else
- Third, [double-check that there's enough space for the LwIP buffers when allocating the output buffer](https://github.com/Aircoookie/ESPAsyncWebServer/blob/2d3af2c32b1dd1a8fd9fb01d7b51fa3653221a89/src/WebResponses.cpp#L301) -- because there's no point in `AsyncAbstractResponse` allocating a large output buffer if the TCP stack can't then allocate a `pbuf` to send.

I did not opt to shrink the buffer to remove the successfully written data, though this could have been implemented as another solution for reducing ongoing memory load.

Unfortunately I'm a bit oversubscribed at the moment and I won't be in a position to PR these changes upstream anytime in the near future (sept-oct time frame maybe).  If anyone else wants to take a stab at it, one key thing to be aware of there is the use of [`class DynamicBuffer`](https://github.com/Aircoookie/ESPAsyncWebServer/blob/2d3af2c32b1dd1a8fd9fb01d7b51fa3653221a89/src/DynamicBuffer.h), which is essentially a shallow replacement for `std::vector<char,default_init_allocator>` that permits allocation failure without exceptions.   (Our project requires graceful handling of memory failures on ESP8266, which doesn't support C++ exceptions, so `std` containers cannot be used.  [`default_init_allocator` is a construct that allows a vector to be `resize()`d without zero-initializing the contents](https://stackoverflow.com/a/21028912/8715474), so it's slightly faster when used for transient buffers.)

mathieucarbou, 2025-07-28:
Thanks @willmmiles !
I was pretty sure you got it handled ;-)

JosePineiro, 2025-08-22:
I will make a PR with this solution, if you agree.

mathieucarbou, 2025-10-19:
@jonny5532 : would you be able to provide a MRE or point to this branch's PR in order to test this fix done by @vortigont ?

https://github.com/ESP32Async/ESPAsyncWebServer/pull/316

Thank you!