ESP32Async/ESPAsyncWebServer #292: Uploading a zero byte file does not fire any write event

Gerporgl, 2025-09-24:
### Platform

ESP32

### IDE / Tooling

pioarduino

### What happened?

When doing a request handling such as _webserver->on("/files", HTTP_ANY, handleFileRequest, handleFileUpload), and sending a zero byte file, the handleFileUpload never gets called.

I would have expected to receive a handler call for handleFileUpload with final==true, index==0, len==0, etc.

This seems a bit awkward to do in the request handler afterward, because of the multipart upload, you can't only check the content-length, and also need to get the filename, etc.
At least if the handleFileUpload was sending a zero len/final event this would allow for a more generic and straightforward file writing implementation.

### Stack Trace

There is no stack trace in this case

### Minimal Reproductible Example (MRE)

Need to upload a zero byte file from a browser page designed for this, or I guess you can do it with curl as well, then implement the handler like this:
```
_webserver->on("/fileupload", HTTP_ANY, handleFileRequest, handleFileUpload);

 void handleFileUpload(AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
       // This never get called
}

void handleFileRequest(AsyncWebServerRequest *request) {
        // This gets called
}
```

### I confirm that:

- [x] I have read the documentation.
- [x] I have searched for similar discussions.
- [x] I have searched for similar issues.
- [x] I have looked at the examples.
- [x] I have upgraded to the lasted version of ESPAsyncWebServer (and AsyncTCP for ESP32).

mathieucarbou, 2025-09-24:
Hum... That's interesting indeed.
I can reproduce with our Upload example:

```c++
  server.on(
    "/upload/binary", HTTP_POST,
    [](AsyncWebServerRequest *request) {
      // response already set ?
      if (request->getResponse()) {
        // 400 No Content-Length
        return;
      }

      // nothing uploaded ?
      if (!request->_tempObject) {
        return request->send(400, "text/plain", "Nothing uploaded");
      }

      uint8_t *buffer = reinterpret_cast<uint8_t *>(request->_tempObject);
      // process the buffer

      delete buffer;
      request->_tempObject = nullptr;

      request->send(200, "text/plain", "OK");
    },
    [](AsyncWebServerRequest *request, String filename, size_t index, uint8_t *data, size_t len, bool final) {
      Serial.printf("Upload[%s]: start=%u, len=%u, final=%d\n", filename.c_str(), index, len, final);

      // first pass ?
      if (!index) {
        size_t size = request->header("Content-Length").toInt();
        if (!size) {
          request->send(400, "text/plain", "No Content-Length");
        } else {
          Serial.printf("Allocating buffer of %u bytes\n", size);
          uint8_t *buffer = new (std::nothrow) uint8_t[size];
          if (!buffer) {
            // not enough memory
            request->abort();
          } else {
            request->_tempObject = buffer;
          }
        }
      }

      if (len) {
        memcpy(reinterpret_cast<uint8_t *>(request->_tempObject) + index, data, len);
      }
    }
  );
```

```bash
~/.../Workspace/me 
❯  rm empty.txt
rm: empty.txt: No such file or directory

~/.../Workspace/me 
❯  touch empty.txt

~/.../Workspace/me 
❯  curl -v -F "data=@empty.txt" http://192.168.4.1/upload/binary
*   Trying 192.168.4.1:80...
* Established connection to 192.168.4.1 (192.168.4.1 port 80) from 192.168.4.2 port 49560 
* using HTTP/1.x
> POST /upload/binary HTTP/1.1
> Host: 192.168.4.1
> User-Agent: curl/8.16.0
> Accept: */*
> Content-Length: 199
> Content-Type: multipart/form-data; boundary=------------------------JFJz2IDemag0uYFop4uoTm
> 
* upload completely sent off: 199 bytes
< HTTP/1.1 400 Bad Request
< Connection: close
< Accept-Ranges: none
< Content-Length: 16
< Content-Type: text/plain
< 
* shutting down connection #0
Nothing uploaded
```

Indeed, the upload handler is not called, but just the request handler, and consequently request->_tempObject is null, which shows "Nothing uploaded".

I agree with you that uploading an empty file should be made possible, and this is a valid use case also.

To me, this looks like a bug.

@me-no-dev @willmmiles : any thoughts ?

Gerporgl, 2025-09-24:
Also, I haven't tried directly myself with the non async "WebServer", but I am doing a migration to AsyncWebServer in a project that already existed, and this was working before, so unless I changed something there, I think this could be the behavior of the default ESP32 WebServer.

closed 2025-09-26 by mathieucarbou

Gerporgl, 2025-09-26:
Confirming this works for me. Thank you!

commit 0ee90902 (2025-09-26): Fix #292: Upload handler was not called when an empty file was uploaded.
Uploading an empty file is a valid use case.
Ref: https://github.com/ESP32Async/ESPAsyncWebServer/issues/292
commit 65bee85b (2025-09-26): Merge pull request #293 from ESP32Async/issue/292

Fix #292: Upload handler was not called when an empty file was uploaded.